prusa3d/Original-Prusa-i3 #119: The current MK3S extruder body is 0.4 mm loose(r)

3Delights, 2019-02-15:
The user **jonnieZG** over on the Prusa Forum is reporting that: _The current MK3S extruder body is 0.4 mm loose(r)_ and that it might be a good idea to change this as it is causing the extruder to wiggle.

Please see this forum topic:
[https://shop.prusa3d.com/forum/general-discussion-announcements-and-releases-f69/the-current-mk3s-extruder-body-is-0-4-mm-loose-r--t27757.html#p128763](
https://shop.prusa3d.com/forum/general-discussion-announcements-and-releases-f69/the-current-mk3s-extruder-body-is-0-4-mm-loose-r--t27757.html#p128763)

jonnieZG, 2019-03-06:
Another user also pointed out that it would be good to change the neck grip tolerances from +0.2 mm to 0.1 mm, so the neck socket diameters would be 16.1 mm and 12.1 mm. I added that to my mod here https://www.thingiverse.com/thing:3428292, but it might be a good idea to make the change on the official model too.

tstoegi, 2020-07-08:
Still no comment or recommendation from Prusa yet?

jonnieZG, 2020-12-10:
This issue is fixed in R6, as all tolerances are set to the values proposed here. R6 introduces also some other improvements, so we can close this issue as solved.