prusa3d/Original-Prusa-i3 #124: Minor tweak to remove "loose end" on Mk3s extruder body.

Mjankor, 2019-03-01:
Not sure if I did this changing nozzle, or whether it happened due to radiant heat, but the loose end of the extruder-cover.stl, near where it meets the Pinda mount drooped down on my (polycarbonate) extruder. 

On the old Mk3 extruder, this loose end was supported by the Pinda mount on the extruder body. It may be worth having a small tab that links to the pinda mount on the extruder body. 
[image]
[image]

Mjankor, 2019-03-02:
I've added it to my Mk3h extruder body. Feel free to incorporate that into the main design. Also, there's a very thin walled section in the centre of that surface between the top cutout and the bottom ridgeline. Might be worth shifting the start of the cutout outwards, away from the e3D, a bit.

Serveurperso, 2019-06-15:
I have this part that systematically melt on the official MK3S parts.
Now I remove entirely this useless lose end.