prusa3d/Original-Prusa-i3 #125: hardware stops heating waiting for inserts... operator asleep 17 hour print... I need to continue the print... how?

carterm2, 2019-03-07:
Prusa I3 Mk3 3.6.0
Latest Slicr3

I am trying to find out how to finish the last 5% of a print after a heater stop... please read to find out the situation.

...
In Slic3r Prusa... I added an M600 to my print, with the plus button....

[image]

I inserted the recommended text, from another Prusa Forum Post...

https://shop.prusa3d.com/forum/others-archive--f82/pause-print-in-gcode-t2034.html#p125758

 in a text editor... inplace of M600

M300 ;ACTIVATE BEEPER
G1 X10.000 Y200.000; parking position GET THE PRUSA TO STICK IT'S TONGUE OUT!
M1; user stop
G28 Y
M105; return to current temp

This stops the print, beeps, moves the extruder to the back left side of the printed, sets up a resume/cancel button on my OctoPrint screen, rehomes Y, and is supposed to....

Return to print temperature.

But I was asleep, and the printer turned of the heat for safety... I forgot the error message on the Prusa Screen

Q1
How can I start the print for the last 5%... Homing Z with the Panda probe will knock the part off the bed...

Q2
Can I replace the M600 text with my own in Slic3r?

Thanks,

Carter

tobymurray, 2019-05-21:
This repo is for the printer parts - I imagine you'll find a bit more traction on https://github.com/prusa3d/Prusa-Firmware (the repo for the software that actually executes on the printer).

Prusa-Support, 2022-05-16:
Since this issue is over 3 years old now and actually doesn't belong to this repo, I will be closing it now.

Marek Micek
Prusa Research

closed 2022-05-16 by Prusa-Support