prusa3d/Original-Prusa-i3 #126: GCodes for MMU2S printed parts reset my printer

acorderob, 2019-03-23:
The gcodes provided at https://www.prusa3d.com/prusa-i3-printable-parts/ for the MMU2 to MMU2S upgrade (for the MK3) reset my printer just after mesh bed leveling.

I'm using FW 3.6.0.

This happened with these gcode files:

- MK3S_extruder_part1_PET_8H30MIN.gcode
- MK3S_extruder_part2_PET_1H30MIN.gcode
- MMU2S_MK3S_S_BUFFER_PET_9H15MM.gcode

This is the start of one of them:

```gcode
M73 P0 R523
M73 Q0 S528
M201 X1000 Y1000 Z1000 E5000 ; sets maximum accelerations, mm/sec^2
M203 X200 Y200 Z12 E120 ; sets maximum feedrates, mm/sec
M204 P1250 R1250 T1250 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X8.00 Y8.00 Z0.40 E1.50 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
M107
M115 U3.5.1 ; tell printer latest fw version
M83  ; extruder relative mode
M104 S230 ; set extruder temp
M140 S85 ; set bed temp
M190 S85 ; wait for bed temp
M109 S230 ; wait for extruder temp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
G1 Y-3.0 F1000.0 ; go outside print area
G92 E0.0
G1 X60.0 E9.0  F1000.0 ; intro line

```
The last thing that i know it runs is the mesh bed leveling (G80). Then it just resets without starting the intro line. I don't see anything wrong with the gcode.

Slicing the stls myself worked fine and i could print them.

thadude3, 2019-06-25:
I am randomly seeing this issue too. Was working fine then just started resetting right at the priming phase.

tdub415, 2019-07-01:
I'm also getting resets after the mesh level when it should be priming on some files.

tdub415, 2019-07-01:
Though it seems to happen only when I'm using gcode sliced in a non prusa slicer, operating in single mode, when filament isn't loaded to nozzle.  The same files that reset when I try running them with filament not loaded to nozzle seem to work fine when filament is loaded to nozzle prior to starting.  I am also printing via Octoprint, rather than SD card.

nnarc, 2019-07-02:
Having the same issue with firmware v3.7.2 using gcode for the prusa enclosure v2 found here: https://www.prusaprinters.org/prints/3673-mmu2s-enclosure

Have a MK3S with MMU2S, using SD.

Project381, 2019-07-12:
I also have this issue, when filament is not preloaded to nozzle the printer will reset after calibration has completed. It happened with v3.7.1 and now with 3.7.2. The printer would also also reset when spool join was turned on and the filament ran out, this seems to have been resolved since updating to 3.7.2.

bornhaeusser, 2019-08-16:
This happens to me too 3.7.2 - i run in this problem since i activated SpoolJoin on.
I deactivated it and maual load it into nossel. Then Print was starting normal.

closed 2023-05-14 by acorderob