prusa3d/Original-Prusa-i3 #127: endstop-block.stl corrupt?

sarf2k4, 2019-04-02:
Hi,
I am unable to load endstop-block.stl and I believe it is corrupted. I'm unable to load into cura 3, slic3rpe, simplify3d, meshmixer and stlviewer as all of them would throw an error

draeath, 2019-05-04:
No such issue here for me, and the file hasn't been updated since Feb 27. I'm able to open the .stl in Slic3r PE, and able to import it into Blender 2.79.

This issue looks to be on your end.

sarf2k4, 2019-05-04:
still the same issue opening the exact file
[image]

bubnikv, 2019-05-04:
would you please attach the file here?

sarf2k4, 2019-05-04:
here it is, downloaded using latest forefox to this writing date
[endstop-block.zip](https://github.com/prusa3d/Original-Prusa-i3/files/3144541/endstop-block.zip)

bubnikv, 2019-05-04:
it is a html document, not an stl.

sarf2k4, 2019-05-04:
Sorry, I was using "save link as" instead of downloading the whole zip, model loaded correctly

closed 2019-05-04 by sarf2k4