prusa3d/Original-Prusa-i3 #131: MMU2S pulley body not strong enough

tobymurray, 2019-04-26:
The selector of the MMU2S currently homes against the "left" (non-motor) side of the [mmu2-pulley-body](https://github.com/prusa3d/Original-Prusa-i3/blob/MMU2/STL/mmu2-pulley-body.stl). As I received it from Prusa, it's simply not strong enough. When the selector homes (by ramming into the side) the left side flexes enough that it causes a misalignment of the selector and the PTFE tubes that feed the selector.
I confirmed this by hand, first by turning off the printer and manually turning the selector screw until the filament could slip through (confirming it was the selector blocking it), and secondly by holding the pulley body upright with my hand when the selector initially homes. Both of these yielded a well-positioned selector.
Not really sure if this is a design flaw or implementation flaw (not printing with enough perimeters to hold it rigid), but either way it is a pain.

Nailig, 2019-05-01:
I seem to have the same problem. I already found remakes on thingiverse which widen the hole towards the finda sensor to try to solve this problem. But perhaps that is not the best solution.

xiphmont, 2019-07-20:
The firmware, as written, is not intended to home to the left.  It's trying to home to the right and watch for a motor stall.  All that back and forth before it goes left (or in some cases just jagged left movement)?  That's it looking for the right stop and not finding because the process is unfortunately buggy and terminating erroneously due to driver stutters.  The 'ramming the left stop' is a result of that.  OTOH, it's also something of a failsafe in that it sort of works, and gets the job done most of the time.

I *think* I know why this is happening, but I need more testing to be sure. Unfortunately, it requires some amount of firmware surgery, so it's a simple fix but not a tiny patch...