prusa3d/Original-Prusa-i3 #132: Missing SCAD files

mariusvw, 2019-05-19:
In the last "delete" round for the MK3S a several parts got removed from the repository.
For example the cable-holder file.

Also I notice that the OpenSCAD / STEP and STL files are completely out of sync.

I already spoke with you guys once that I am willing to spend some time to organise this but I must be sure to have the latest production revision.
Also the STL generation can be automated. Even with the self-print / prusa-print part numbers within the files.

stuartpb, 2019-05-30:
I think the cable-holders are generated from modules in the Einsy-base.scad file.

What *I* can't find, however, is the code for the Raspberry Pi Zero frame. There's a comment that says "Raspberry frame" in Einsy-base.scad, but there's nothing to do with the frame underneath it, as if the frame was drafted in this part of the file, but the code was moved to another file that wasn't added to the commit.

maximlevitsky, 2019-07-06:
I confirm that.

mariusvw, 2019-07-06:
> I think the cable-holders are generated from modules in the Einsy-base.scad file.
>
> What _I_ can't find, however, is the code for the Raspberry Pi Zero frame. There's [a comment that says "raspberry [...]

The cable-holder.scad is the part on the back of the print head, not the two clamps of the Einsy-base.

Not sure but I get no clear response from Prusa how we can manage this repository together so soon I will be releasing a re-organised version, don't think it won't be merged but it will help working faster a lot.