prusa3d/Original-Prusa-i3 #133: Hotend wires get damaged due wrong cable-holder

mariusvw, 2019-05-19:
Lately I got a several MK3 upgrades in service, it happens they all run into the problem where the hotend wires get too low and hit the heatbed cover bolt where the bolt-end is facing up.

[image]

Here someone already started a discussion about this with a good photo showing what happens.

https://www.reddit.com/r/3Dprinting/comments/b9dxhc/just_noticed_that_the_red_wire_coming_out_of_the/

Again, I am willing to help developing on improvements. Just need the right files and workflow.

austinvojta, 2019-05-19:
Simple modification, I have changed the heatbed cable cover to use the problematic screw flipped around. So now you put a nut on the underside, screw goes in from top-down, and sits counterbored inside the printed part to avoid the sharp end of the bolt sticking in the air and causing this problem. SCAD and STL files linked below.

https://drive.google.com/drive/folders/1wAVW9g4GOHe4tTdY2j6lgEapmjQKBeSA?usp=sharing

mariusvw, 2019-05-20:
That is one part of the solution, will add the second solution to the cable holder later this evening.

Side-question, isn't it better to add the files in a pull request instead of Google Docs?

austinvojta, 2019-05-20:
Probably so! I have never made a full request before, don't know how to do
it. Also I think Prusa has reasoning to leave the screw entering from the
bottom - it leaves less chance to create a short circuit accidentally
between contacts on the bed if someone were inserting/removing screw while
bed is powered. So not sure they want to change the file.

akukan, 2019-05-21:
We will go with similar solution as @austinvojta is suggesting, this is currently finishing tests and once goes to production it will be updated here - matter of days. It's already covered here atm https://manual.prusa3d.com/Guide/7.+Heatbed++&+PSU+assembly+(Black+PSU)/1153#s26163

mariusvw, 2019-05-21:
This solves the heatbed issue but are there also working plans on the Xaxis body/cable guide? Cause this is actually the reason why the wires get ~1.5mm lower than in the previous version.
Since the cable holder got removed from the repository I did not started on updating this part.

Another thing for the heatbed part is if you'd like I'd want to add in something to prevent short circuiting the heatbed wires.

gregsaun, 2019-05-21:
@akukan there is a risk that, during maintenance, someone shortcut + and - of the heated bed with the wrench. Maybe you can keep keep your previous solution and add a washer on the screw head.

Panayiotis-git, 2019-05-21:
[image]

gregsaun, 2019-05-21:
:satisfied: you know how that goes, we all went there I think... There is a safer way so why taking the risk?

mariusvw, 2019-05-22:
As part as a solution to this without reprinting the heatbed cover yet till feedback from Prusa I restored and adjusted the cable-holder.

There is an extra cable tie and angled cut-off close to the body to keep the wires up.

[image]

rrauenza, 2019-05-25:
@mariusvw Do you need help doing a pull request?

mariusvw, 2019-05-25:
@rrauenza no help needed, just didn't made one since my other PR's are not getting processed.

rrauenza, 2019-05-28:
> @rrauenza no help needed, just didn't made one since my other PR's are not getting processed.

For what it's worth, I did print yours out for my upcoming upgrade...  Thanks for the change.  You might also put it on thingiverse?

ManuGithubSteam, 2019-07-07:
Semms like a good improvement :-)

mariusvw, 2019-07-07:
> > @rrauenza no help needed, just didn't made one since my other PR's are not getting processed.
> 
> For what it's worth, I did print yours out for my upcoming upgrade... Thanks for the change. You might also put it on thingiverse?

Uploaded here: https://www.thingiverse.com/thing:3733759

draeath, 2019-12-02:
I think this was fixed via e55953ff5189ca9b109a8d779e8a57ee6fd271e0 ?

draeath, 2019-12-02:
@akukan looks like you did in fact merge that change in, we can probably zap this issue!

e55953ff5189ca9b109a8d779e8a57ee6fd271e0

mariusvw, 2019-12-03:
@draeath, @akukan
Indeed look like it has been added... Just wondering, are you guys moving away from OpenSCAD?

Asking this because I see that the SCAD files are not matching with the current STL versions or the SCAD file is completely removed.