prusa3d/Original-Prusa-i3 #134: Half of print worked well, but then the print shut down when color change was needed. Why?

ErrorlessJinX, 2019-05-22:
Currently, I'm working in an internship where I'm working heavily with Prusa 3D printers. The primary printers I've been working with so far is the Prusa i3 MK3S. I've been slowly getting some things dialed in on it, but had an issue come up that no one seems to understand why it happened.

The first 3/4 of the print went great. However, there was a color change needed around that point but I had been in a meeting for a while during that time. The printer head had moved to its normal position for the material change, but when I got to the printer, the head and pad had both completely cooled down and the printer didn't show as having any print in the queue. Will the print time-out if the material is not changed soon enough? Or is there something else going on that I can't try to fix?

I'd like to fix this issue because my prints are in the range of 24 to 28 hours for each print. It would be nice to not have the prints fail after getting 20 hours into a print. I will submit any info if you need it, but I'm assuming it is just something with the printer overall because it has done fine with plenty of other prints using the same settings, material, etc.

stahlfabrik, 2019-05-22:
This repository is about the hardware of the printer - especially printed parts. I would advise to post the issue again on the firmware repository

ErrorlessJinX, 2019-05-22:
> This repository is about the hardware of the printer - especially printed parts. I would advise to post the issue again on the firmware repository

Okay. May I ask how? I'm not particularly familiar with forums and such. Just looking for a little help from the community!

stahlfabrik, 2019-05-22:
Sure.

If you want to address mostly the community, you could ask in the official forum. https://www.prusaprinters.org

But since you identified an issue, you did the right thing and opened an issue, here, on Prusa's GitHub. You just created the issue in the wrong repository. The right one would be this: https://github.com/prusa3d/Prusa-Firmware/issues

stahlfabrik, 2019-05-22:
Oh and I should add: I think cooling down of the hotend is expected, but the heatbed should NOT cool down while the print is paused. That is why I would say, that this sounds like a bug in the firmware.

ErrorlessJinX, 2019-05-22:
Awesome, thank you so much for the help!

scalda, 2019-11-20:
Hi @ErrorlessJinX did you ever post this elsewhere and did you get a fix for it, as I am having the same issue and can't find anything other than this thread?

Regards
Scalda