prusa3d/Original-Prusa-i3 #135: MK3S: Unable to back off the filament on filament runout

maximlevitsky, 2019-07-06:
When filament runs out, it is often quite curled due to being on the inner diameter of the spool.
When the IR sensor detects the runout, user is asked to unload the filament but when this is attempted, the filament is stuck inside 'adapter-printer.stl' because it is a bit curled around the ball, and thus it doesn't hit the hole through which it was inserted in first place.

I manually fixed that by enlarging that hole, and adding a chamfer to the corresponding hole in the filament sensor cover from below.

ManuGithubSteam, 2019-07-07:
Hi
This seems like a good upgrade :-) Would you share your stls or design files ?
Thanks

maximlevitsky, 2019-07-07:
I did that in the old fashoned way, with a knife :)

tkircher, 2019-07-12:
I had this exact issue as well. Essentially every time I have a filament runout, I have to unscrew the idler and back the filament out with pliers. Don't really want to disassemble the hotend to fix it, since it happens so infrequently, but Prusa should really fix this.

PeterWelzien, 2020-04-06:
I also have this exact issue EVERY time I run out of filament. It's quite annoying having to open the idler and trying to get a hold of the filament with some pliers and pull them out. A fix would be much appreciated.

DWave42, 2020-04-14:
I am starting a chat with Prusa right now about this issue.

acetech09, 2020-04-18:
I've also had this issue every time I run out of filament, since I assembled the machine. I saw this thread a while ago, tried to chamfer/open up the holes, it still occurs.

tkircher, 2020-04-18:
The right solution is probably to move the head up and to the home position and then extrude out the rest of the filament, rather than trying to unload it.

bartlee01, 2020-08-17:
I have this same problem on Mk3S and Mk3S with MMU2S. I have been dealing with it by straightening or cutting off the kinked end of the filament but this program needs fixing. Another fix would be for the end bit of kinked filament to be much longer and have a rounded tip.

tkircher, 2020-08-17:
@akukan @josefprusa @veverkaondrej Still looking for a workaround on this issue. It's still an issue.

TurboCooler, 2021-01-04:
@akukan @josefprusa @veverkaondrej -- just lost a 20hr print because of this issue.   After the filament ran out, it was not able to retract the remaining filament.  I opened the door to remove the filament stuck in the extruder.  While doing so the X moved as the filament was really jammed.  The print did not recover and I lost the print.   A very SIMPLE solution would be to add code to firmware that when the filament runs out that the X/Y move to a corner and simply purges the remaining filament.  You have the signal that filament has runout.  You know the length that is left based on the height from the gear to the sensor.  Simply extrude that amount and allow someone to load additional filament and continue.   This would eliminate the problem completely.

bartlee01, 2021-01-04:
When filament runs out on mmu2s this problem is even worse. Unless you have manually straightened and cut a rounded point on the filament end it almost always gets stuck in the mmu2s on unload. It should park in a corner and give you the choice to either unload or purge the remaining filament.

tkircher, 2021-01-04:
@TurboCooler @bartlee01 Agreed that there should be an option to purge out the filament. Part of the problem is that the extruder gear can't drive the whole dead volume out of the extruder by itself, which is why I suspect they did it this way in the first place. But it's definitely superior to purge out the old filament with the new.

3d-gussner, 2021-01-05:
Hello everyone,
Nice to see some discussion here, but as this repository is related to the "**hardware**" design and not the firmware I think it is in the wrong place.
Please reopen a new issue [here](https://github.com/prusa3d/Prusa-Firmware/issues) if you think it is firmware related and close this issue here.

tkircher, 2021-01-06:
@3d-gussner This is still a hardware issue, but it may also be a firmware issue, or it's possible that the hardware can't be fixed and there can only be a workaround in firmware. TBD.

olilam, 2021-02-16:
Exact same issue on my last print! A fix would be great!

TheLion, 2021-03-28:
Just had the issue on 2 consecutive prints on my MK3S. I hadn't figured out the issue, but When reading te comments, I now know 😜. 

I do wonder if the MK3S+ upgrade fixes this issue, as I already have the kit lying around 😉

Does anybody know if this is the case?

TurboCooler, 2021-03-28:
> Just had the issue on 2 consecutive prints on my MK3S. I hadn't figured out the issue, but When reading te comments, I now know 😜.
> 
> I do wonder if the MK3S+ upgrade fixes this issue, as I already have the kit lying around 😉
> 
> Does anybody know if this is the case?

No, it does not.  The only thing that works provided you can catch it on time is to cut the end at 45 degrees so there is no hook or blob on filament on the end.  Then using your fingers make the end as straight as possible.  This works for me but it is a pain to time on very long prints. 

The worst part of this problem is if you get a jam and while clearing it you move the head or plate even a tiny bit it will never be able to restart correctly.  If they homed the head or somehow reset where it left off it could work if you get a jam.  Glad to see I am not the only one with this issue and hopefully there is a fix.

tkircher, 2021-03-28:
Unfortunately, limit switches (or collision detection in the case of the Trinamic drivers) aren't precise enough for reliable re-homing. There just needs to be a better way to back out the filament and remove jams. Though, the MK3S+ did include some extruder tweaks to improve the situation.

UltraSalem, 2021-10-03:
It's been a while since I used it, but I seem to recall my mk2s runout sensor worked fine and the filament would back out and let me reload new filament without issue. Not sure what changed inside the extruder block for the mk3s to break it, but it jams backing out every time on my mk3s

napter, 2021-10-13:
Does anyone have a good solution to this other than snipping the filament end?  I was thinking of designing a mod to move the sensor up and out of the extruder body so that there is a gap where you can pull the filament out. I made a similar setup on my ender 5 but that sensor was separate to begin with.

lifewithmy3Dprinter, 2023-12-06:
I'm having the same issue. The filament gets stuck when trying to back out.

Three fixes:

- Ability to choose not to back it out, but rather purge it out while pushing new filament behind it.
- Add a chamfer or a wider hole for the filament to back out of.
- Do a homing routine after a new filament is ready to go. It's frustrating to loose a print because you accidentally move the extruder trying to get the stuck filament out.

ArmoredCarbon, 2024-07-07:
I am having this issue as well, and I always clip the bent ends of the filament and straighten it.
Has anyone come up with an easy solution yet?
If not I'm just going to rebuild the runout sensor which I know is going to be a complete pain.
My current idea is to drill it out and run a ptfe tube through the sensor assembly. A hole in the side of the tube should allow the sensor to detect the filament.

napter, 2024-07-15:
> I am having this issue as well, and I always clip the bent ends of the filament and straighten it. Has anyone come up with an easy solution yet? If not I'm just going to rebuild the runout sensor [...]

I made a very crude mod that takes the original sensor housing and puts it on a leg so it sits above the extruder. Since there is a gap between the sensor and the extruder, the bent end never goes into the extruder (and can be pulled out of the sensor if it didn't exit completely).