prusa3d/Original-Prusa-i3 #136: PINDA holder sags in R4 revision

lff12, 2019-07-16:
After a lot of high temperature printing (Polymaker PC Max and generic HIPS), the PINDA holder begins to sag slightly, causing the sensor to tilt slightly to the left, and decreasing the accuracy of readings. Reseating the sensor and recalibrating the printer helps, but there is still a distinct sag. Parts for R4 self printed in Innofil PETG. Is it possible to reinforce the PINDA holder to prevent this?

prusa3d-bb, 2019-07-18:
Hello,
do you have any photos of this, which I could send to plastic parts developers? We are already reinforcing some parts, but I am not sure if this one is on the list too

Best Regards / S pozdravem
—
Jakub Kučera

Customer Support

_______________________________________

PRUSA Research

188/7a Partyzánská, 17000, Prague

+420 222 263 718

+421 220 570 305

Troubleshooting tutorials here!

Follow us on twitter and Facebook! 

Check out our blog!

________________________________________

Please rate our reply ...be generous :-)

--- original message ---

--- end of original message ---

lff12, 2019-07-19:
[image]

lff12, 2019-07-19:
Sorry about the quality - i had a print in progress - but you can see that the PINDA is at an angle - its not caused by the position of the cable - it tilts to the right even if the cable is moved elsewhere.

prusa3d-bb, 2019-07-19:
Hello,
thank you for the photo. It is visible there. We are already reinforcing the parts around pinda probe. Do you need a replacement plastic part or can you print it yourself?

Best Regards / S pozdravem
—
Jakub Kučera

Customer Support

_______________________________________

PRUSA Research

188/7a Partyzánská, 17000, Prague

+420 222 263 718

+421 220 570 305

Troubleshooting tutorials here!

Follow us on twitter and Facebook! 

Check out our blog!

________________________________________

Please rate our reply ...be generous :-)

--- original message ---

lff12, 2019-07-21:
Very happy to print it myself, thanks! It also seems to be touching the cover and melting there - seems there is less gap between the heatsink and bottom of the cover than previous version! Thanks though.

lff12, 2019-07-27:
[image]
Here is a few photos of the old body after removal. Interestingly the fan part is perfect, it's just the body and cover that deformed. I'm replacing the parts today so will report back

lff12, 2019-07-27:
[image]

prusa3d-bb, 2019-07-29:
Hello,
thank you for the photos. The reinforced parts are done. You can print them. https://github.com/prusa3d/Original-Prusa-i3/tree/MK3S/Printed-Parts/STL​

Best Regards / S pozdravem
—
Jakub Kučera

Customer Support

_______________________________________

PRUSA Research

188/7a Partyzánská, 17000, Prague

+420 222 263 718

+421 220 570 305

Troubleshooting tutorials here!

Follow us on twitter and Facebook! 

Check out our blog!

________________________________________

Please rate our reply ...be generous :-)

--- original message ---

maximlevitsky, 2019-07-29:
Thank you!
Note that in my case the extruder cover melted right away after I preheated the printer to just PETG temperatures. 
PINDA seems to hold up, though.

lff12, 2019-07-29:
I've printed it, twice. First one had a small crack around the PINDA holder - but around the area where the screw and nut is so still holds ok - printed a second one. Have run a few prints with temps up to 270C - so far so good - will see in about 6 weeks if its still staying straight.

prusa3d-bb, 2019-07-30:
Hello,
if anything happens. please let us know, so we could reinforce it even more. Thank you

Best Regards / S pozdravem
—
Jakub Kučera

Customer Support

_______________________________________

PRUSA Research

188/7a Partyzánská, 17000, Prague

+420 222 263 718

+421 220 570 305

Troubleshooting tutorials here!

Follow us on twitter and Facebook! 

Check out our blog!

________________________________________

Please rate our reply ...be generous :-)

--- original message ---

prusa3d-bb, 2019-07-31:
Hello,
could you please send me a photo of the new reinforced hotend cover, which was melted already? So it really melted after first heat up?

Best Regards / S pozdravem
—
Jakub Kučera

Customer Support

_______________________________________

PRUSA Research

188/7a Partyzánská, 17000, Prague

+420 222 263 718

+421 220 570 305

Troubleshooting tutorials here!

Follow us on twitter and Facebook! 

Check out our blog!

________________________________________

Please rate our reply ...be generous :-)

--- original message ---
On July 30, 2019, 5:44 PM GMT+2 info@prusa3d.com wrote:

Hello,

if anything happens. please let us know, so we could reinforce it even more. Thank you

Best Regards / S pozdravem

—

Jakub Kučera

Customer Support

_______________________________________

PRUSA Research

188/7a Partyzánská, 17000, Prague

+420 222 263 718

+421 220 570 305

Troubleshooting tutorials here!

Follow us on twitter and Facebook! 

Check out our blog!

________________________________________

Please rate our reply ...be generous :-)

> I've printed it, twice. First one had a small crack around the PINDA holder - but around the area where the screw and nut is so still holds ok - printed a second one. Have run a few prints with temps up to 270C - so far so good - will see in about 6 weeks if its still staying straight.

--- end of original message ---

maximlevitsky, 2019-07-31:
@prusa3d-bb - no no, I haven't yet printed the reinforced hotend cover.
The old one, from the MK3S update melted right away.

I just wanted to note that in my case it melted without using higher that usual temperature, like needed for PC or so.

maximlevitsky, 2019-07-31:
I printed the update myself using the supplied PETG, and official gcode. Still I suspect taht my printer might be a bit off in term of dimisions, so maybe the cover literaly touched the heater block, or even more likely its because I use the e3d sock which might cause the touching.

maximlevitsky, 2019-07-31:
Also it probably helped that ambients are here around 30C. (I basically have two choices: suffer from AC noise, or suffer from heat... )

Moorviper, 2019-09-11:
Why not use just ABS for the extruder part.
Many PETG are not better than PLA and deform when warmed up to 55-60°C for a longer period.

francisdb, 2019-09-15:
Same problem here since I built the [prusa lack enclosure](https://blog.prusaprinters.org/cheap-simple-3d-printer-enclosure/). I print a lot of PC / Nylon at high temperatures and the ambient temp in the enclosure reaches 35-40°C. The pinda probe started sagging and had to re-calibrate 1st layer twice already.

For reference:
https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-hardware-firmware-and-software-help/mk3s-part-melted/

Has the pinda probe part also been reinforced or only the cover? Any suggestions for a better material than PETG for this part?

prusa3d-bb, 2019-09-18:
Hello,
pinda probe holder has been also reinforced as you can see here https://github.com/prusa3d/Original-Prusa-i3/tree/MK3S/Printed-Parts/STL
I recommend using PETG or ABS/ASA. We just released Prusament ASA, which you can use. 

Best Regards / S pozdravem
—
Jakub Kučera

Customer Support

_______________________________________

PRUSA Research

188/7a Partyzánská, 17000, Prague

+420 222 263 718

+421 220 570 305

Troubleshooting tutorials here!

Follow us on twitter and Facebook! 

Check out our blog!

________________________________________

Please rate our reply ...be generous :-)

--- original message ---

https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-hardware-firmware-and-software-help/mk3s-part-melted/

embusalacchi, 2019-10-07:
Hi!  Just wanted to report that my original hot end cover has started to sag with only PETG printing.  I was printing the replacement in PETG so I'm glad that's when it started to sag.  I'm not seeing it with the PINDA holder though.  I plan on replacing the cover in the next few days assuming I get the time.  I can report back if it sags again.

Rubines, 2020-02-03:
Hi

I also had thes sagging pinda problem and ignored it, until the probe holder broke,..
The problematic section is where the square nut is positionend, there is verry little supporting structure around it. Unfortunately I cannot find the improved version anymore, the R4 Version started to sag pretty quickly.
[image]

prusa3d-bb, 2020-02-04:
Hello,
please contact Prusa support directly on the email info@prusa3d.com or on the online chat, which is located on the eshop bottom right corner to sort out the replacement of this part. Thank you

wjsteele, 2020-04-07:
Just as an FYI... I just experienced this same failure during an extended print at 250C (Prusament PETG) where the Pinda probe sagged and came in contact with the print.  I was able to pause the print and move it up, but will need to repair it before the next print.

rafepurnell, 2020-04-26:
I have had the same issue and have melted 3 of these reinforced PINDA holders. I'm starting to worry about finding a solution. 

Right now my CR10 is my savoir. 

Is the right path to get the BondTech upgrade or is there another solution for this problem.

I loved my MK3 before the MK3S upgrade. :(

observatoer, 2020-04-26:
Based on what I have tried, the Bondtech upgrade extruder and the Bear extruder have both no issue with the pinda holder. You shall to run them with a Sunon fan to cool the hotted properly.

Rubines, 2020-04-28:
Hi

The new design update solved all my troubles even when printing @250 °C for 18 hours.
At first I was sceptic but after a while all my concerns were gone.

Prusa-Support, 2022-05-10:
Hello! Sorry for the late response to this thread. Closing the thread as the reinforcing aspect is solved - of course, in case of any further specific issues related with this, do not hesitate to let us know by contacting our Technical Support info@prusa3d.com at any time, we'll help with anything and check each case one by one. 

Alessandro Pantaleo
Prusa Research

closed 2022-05-10 by Prusa-Support