prusa3d/Original-Prusa-i3 #138: MK3S - Random print fail partway through

tmfksoft, 2019-09-21:
I've just received my MK3S aand assembled it.

But partway through a print, at random it suddenly raises the Z axis, the print bed goes all the way to the back and repeatedly goes forward a few times then the X axis flies off to the right and jams.
After is done what I can only refer to as a stroke it goes back to printing but the X axis is now offset by half the print bed size and tries to print on the side of the print bed.

Here's a badly recorded clip:
https://files.minio.thomas-edwards.me/2019/09/21/2019-09-21_23-56-08.webm

Here's the GCode:
https://files.minio.thomas-edwards.me/2019/09/21/2019-09-21_23-57-47.gcode

Generated using Cura 4.2.1 using the Prusa Printer and Profiles.

You can see it printing normally and properly.
Then randomly it'll do its stroke thing then repeatedly raising up.
This happens at random. I've tried the same print twice with it doing it both times at different points within the print.

I can only assume its a firmware thing due to it doing it at random. I've had a few successful prints without this happening but I can't take my eyes off the printer incase it does it.

Panayiotis-git, 2019-09-22:
It looks like your X axis is very tight or the extruder cannot move smoothly. If I understand it correctly, the printer detects a crash. It tries to re-home and it fails to move the X-axis to the very left position.

Are you sure that your X-axis has been assembled properly? Try to remove the belt and move the extruder from the left edge to the rightmost edge and back. You should not feel any resistance. If you feel like you have to push the extruder to move it, then try to un-tight a little the back carriage screws.

tmfksoft, 2019-09-22:
After some bothering others on Twitter I got the same suggestion.
I loosened the belt until the belt status read 265 instead of 255.

After that I had much more success, however it's failed once again 9hrs into a 9hr 42min print which is gutting. The print has come out absolutely perfect up until it failed.

I'm really not sure what's causing this currently. I'll have to try disabling crash detection and see how the prints go afterwards.

tmfksoft, 2019-10-06:
Forgot to update this issue.

After tightening the X belt until it was within the recommended range I was able to print for longer until crash detection kicked in once again.

I disabled crash detection and have made many prints since then, all prints have turned out perfectly fine.

I dont understand what constitutes as a crash for the crash detection to kick in but I've printed many hours of prints since disabling it without issue.

I can only assume the crash detection is over sensitive?