prusa3d/Original-Prusa-i3 #139: Missing SCAD files for MK3s extruder

hillct, 2019-11-03:
It appears many of the MK3S extruder printed part OpenSCAD files are missing from the repository (though STL files are present) including the extruder-body.scad, extruder-cover.scad and extruder-motor-plate.scad among others (IR sensor cover, etc)

These expected files are missing from the MK3s branch, where we'd expect to find such files for the MK3S extruder. If they're located elsewhere in a publically accessible location, please advise, and if not, please update the repository with the missing files. Thanks.

Moorviper, 2019-11-05:
the files are released

hillct, 2019-11-05:
Having reviewed the repository again just now, the files have not yet been released in the format native to the cad package in which they were designed. That is what's meant by 'original cad' files. STEP format is not what is meant here. STEP is a portability format that lacks the data granularity available in the format native to the design package used for these parts.

3d-gussner, 2019-11-12:
@hillct Prusa is so kind to provide the STEP files which can be imported by lot of different CAD programs and so not forcing people to use THEIR choice of CAD software.

You mentioned that STEP may lack of data granularity, didn't see it yet. What are you missing?
What if their CAD format isn't compatible with your CAD software?

Look around and tell me how many 3D printing companies share "real" CAD files and I mean also STEP files. Most just give you STL files.
Some like E3d kindly provide their motion system and tool changer in STEP files but for other designs (hotends) they kindly provide 2D drawings.
Zaribo shares their designs in STEP files, what is awesome as well.
Bondtech shares their designs in STEP files, again awesome that they doing it.

Only 3d Printing company sharing their "raw" CAD files I know of is BCN3D.
But not everyone has Solid Works and sometimes importing other "raw" CAD files into the CAD software of your choice can be frustrating or even be impossible.

STEP can be imported in most CAD software and is acknowledged by all big/small CAD software as the way to exchange CAD information. 

We should be glad that Prusa, E3d, Zaribo, BCN3D,Bondtech .... share more than just STL files.

gregsaun, 2019-11-12:
@3d-gussner hey you forget about me :-D , I do provide Fusion 360 CAD files. Lulzbot does it too (FreeCad) and probably others!

However, I don't think it is very important to compare what others do but what is the idea behind open source. What you want is to share knowledge to innovate faster and better. STEP are missing a lot of information and probably the most important ones (like why this is at that size, the relation between two distances, the motion, etc.). Fusion360, FreeCAD, Solidworks are powerful tools because they are parametric. If you share STEP only you remove all the "parametric" part of the design.

Using a closed or proprietary software is not a problem in open source, what you need is to share the source.

Now you can see STEP as a source to build the printer (even if this is the output of the source used to design the printer). But I don't think it really match goal of open source. Note this is different for a figurine, but on functional or mechanical parts is another level.

3d-gussner, 2019-11-14:
@gregsaun How could I forget you... Didn't mention you as Bear Upgrade isn't a company (right)? And BTW thanks for sharing your Fusion 360 files on Thingiverse and Github. 

I get what you mean and getting "raw" CAD files is nice but getting STEP files is already a "good" way to share designs in my opinion.

I imported the STEP files in Fusion 360 and could modify/measure/analyse the parts.

P.s.: Just downloaded your Fusion 360 files and I must say, that it is nice/easier to follow your design choices. Not sure everyone is willing to share the hole design history :-)

gregsaun, 2019-11-14:
@3d-gussner so sad you forget me :-D . You are right we are not a company.

I agree for STEP, loads easier than STL. However, still not the source to me, it is the output of the source used to design the printer. It is a bit like a compiled software that has been decompiled. You can use it but...

Note that Prusa was/is sharing OpenSCAD files which correspond to CAD files and not to STEP.

Happy you like my files. Regularly people come to me and told me they learned a lot by looking at my files. That is where the force of open source is and why I am not a fan of STEP (yet better than STL sure).

henrikkoch, 2019-11-14:
So where do i find the CAD source files (not the STL's) :-)

PatB42, 2019-12-11:
yes where and the last fines for MK3 S please

infosisio, 2020-05-08:
Very sad that Prusa moved away from OpenSCAD. Proprietary formats are not the right fit.

lf-, 2020-05-09:
@infosisio proprietary formats are just how it is in industry and where the skills are. What's frustrating about this move moreso is that the extruder is now decidedly not open source.

infosisio, 2020-05-09:
@lf- The designs were in OpenSCAD for so long and everything was fine. Don't fall in the trap of what the "industry" is doing. Do what you must.

gregsaun, 2020-05-09:
If you do open source you don't necessarily need do it with an open source software. For multiple reason, have OpenSCAD can be very restrictive and slow to use. It also only support a mesh format which comes with several other limitation (but there are other languages doing the same thing that can export a STEP for example).

MK3S is under GPLv2 which request to share the source code and it sounds like this not the case here. It would be nice to have someone more competent than me to confirm this. And an answer from Prusa anyway.

lf-, 2020-05-09:
@gregsaun I regret bringing up this discourse again because it explodes my inbox and accomplishes nothing of value. I believe in these principles but yelling at each other on GitHub isn't going to get Prusa to change their business practices discouraging derivatives they've been following for years. Have you ever noticed there is no bill of materials for any Prusa printer from the very beginning of them selling kits? That's not a coincidence. 

Anyway, the GPL does not compel the copyright owner to do anything whatsoever: it only requires things of licensees. So there is absolutely nothing legally wrong with them not publishing source unless they are not the sole copyright owners.

I shall now mute this thread. Have a good day.

gregsaun, 2020-05-09:
@lf- Thanks!

hillct, 2020-05-10:
It's not clear to me what the point of @lf- 's response is here. Nobody is accusing @prusa of doing anything legally wrong here. We're just very nicely asking that the files be released in a more robustly editable format that facilitates modification and  community contribution. Alternatively, they could outright state that they have no indent to do so, clarifying for us that they're just interested in the PR benefit of Open Source, rather than the spirit of it. In which a case we could all move over to Lulzbot which has more affirmatively demonstrated their commitment to the spirit of Open Source.

Again, to be clear, nobody is asserting that @prusa is doing anything legally wrong, We're just asking them very nicely to live up to the spirit of the movement from which they've benefitted so very greatly.

gregsaun, 2020-05-10:
For those that search an i3 design fully open source (with CAD files) you can have a look to my projects :
Frame: https://github.com/gregsaun/prusa_i3_bear_upgrade
Extruder: https://github.com/gregsaun/bear_extruder_and_x_axis

infosisio, 2020-05-10:
@gregsaun Thanks though that still won't do as we don't get the parametric control that a freedom respecting software like OpenSCAD would give us.

infosisio, 2020-05-10:
@hillct Very well said!

gregsaun, 2020-05-10:
@infosisio note that you have parametric control in CAD software (variables, math, constraints, projections...) but not as accessible as OpenSCAD code.

infosisio, 2020-05-10:
@gregsaun Yes but I have no interest in gaining that control through a proprietary application. Thanks for sharing what you made anyhow though.

gregsaun, 2020-05-10:
I do understand and respect your point of view. It would be impossible to me to create my parts in OpenSCAD because it would take me too much time. Plus I am not fan of STL/meshes because hard to work on them (not everybody knows how to code). However, I would love to see how my parts could be made on CadQuery or similar.

3d-gussner, 2020-05-10:
@infosisio Why don't you try to redesign one of the complex parts in OpenSCAD. I am curious if you can can achieve this and how long it takes to do that.
The community is always welcome to attribute to the project and if you can provide these openSCAD files others may be interested in these.

infosisio, 2020-05-10:
@3d-gussner I designed all the parts of Hypercube from scratch in OpenSCAD just a few days ago. It is something I can do and that I am willing to do, however our conversation here is not about what I make but why Prusa suddenly decided to skip developing some crucial parts in the old fashioned open way.

brettvitaz, 2020-05-10:
The reality is that the design will be created in what ever format the author is most comfortable with. Demanding OpenSCAD makes no sense to me; why are you trying to dictate the language the authors use to create? Just because you prefer it does not mean others do, and if the author is not comfortable in the language, you will get inferior results. If the authors are living up to their promise of open source, then I have no complaints.

infosisio, 2020-05-10:
@brettvitaz You are taking things out of context. Have you seen what 99% of the parts of the project are already designed in? It's not because I randomly prefer OpenSCAD. It's because that was the initial project promise.

brettvitaz, 2020-05-10:
Where was it promised that everything created would be in OpenSCAD?

infosisio, 2020-05-10:
@brettvitaz The company was (and expect to still be) full force behind OpenSCAD, even creating tutorials and encouraging its users to get familiar with in because that is what the project uses. Please check some facts before asking questions or trying to make absurd points.

brettvitaz, 2020-05-10:
I see, it was not promised. 

I'm with @lf- , this whole thread is absurd and people think they're entitled to something that they are not. @infosisio, you are attacking me for having an opinion that doesn't agree with yours. I'm done with this thread as well.

infosisio, 2020-05-10:
@brettvitaz It was promised and nobody attacked you. Scroll up and see who started being aggressive with non-sensical arguments like "everyone is free to use what software they want, mind your own business". I won't be responding to you from now and on, sorry.

erikmartino, 2020-06-26:
The CAD files are [GPL2](https://github.com/prusa3d/Original-Prusa-i3/blob/MK2/LICENSE). GPL is usually used in software., but he GPL FAQ mentions [how to apply it to non software](https://www.gnu.org/licenses/gpl-faq.html#GPLOtherThanSoftware).

> You can apply the GPL to any kind of work, as long as it is clear what constitutes the “source code” for the work. The GPL defines this as the preferred form of the work for making changes in it.

STEP (Standard for the Exchange of Product model data) The format fits the purpose here very well. However it is not really a format for editing, but for exchange. So according to the interpretation in the GPL FAQ the current state is not optimal.

gregsaun, 2020-06-26:
The problem is the same for the Prusa Mini and someone reported that the STEP files header were showing that Prusa is using Autodesk Inventor.
https://github.com/prusa3d/Original-Prusa-MINI/issues/21#issuecomment-647568789