prusa3d/Original-Prusa-i3 #143: [Proposal] FABUI

infinity0n3, 2020-01-21:
I'm making a new UI for 3d printers and CNC machines.
It's an highly optimized printer control running on raspberry pi with a nice web interface.

There is an initial support for i3 MK2 printers.

You can check it out [on GitLab](https://gitlab.com/colibri-embedded/fabui)

MK2.5 and MK3/S are in the list but have not been tested well yet so MK2 is the suggested printer to try it out on.

origel, 2020-01-22:
Great! can you show me some screen shots?

infinity0n3, 2020-01-22:
Sure, there are [screenshots](https://gitlab.com/colibri-embedded/fabui/fabui-frontend/-/wikis/screenshots) now on the wiki page.

tracernz, 2020-01-23:
That looks excellent! The obvious missing thing is webcam support. Do you have any plans in that area?

infinity0n3, 2020-01-28:
@tracernz There are several things planned after the alpha phase is finished. Like camera support, gcode preview. For now the focus is to get everything working and on multiple printers and to add milling support and grbl gcode flavor support.

closed 2022-05-16 by Prusa-Support