prusa3d/Original-Prusa-i3 #147: [Question] Is READMEmd outdated?

jarl-dk, 2020-04-24:
In the README to this repo it says:

> Original Prusa i3 MK2
> 
> Original Prusa i3 MK2 is a 3D printer project maintained by PRUSA RESEARCH. Originates in RepRap project.

But I belive the parts ar for i3MK3-series.

And link to the product links to MK2, but is redirected to MK3

Is the README outdate? The parts are good for i3MK3, right?

jarl-dk, 2020-04-24:
Ahh. I see, it seems like there is a branch for each printer model...

tracernz, 2020-04-24:
Yes. Maybe it would be a good idea to make MK3S the default branch.

jarl-dk, 2020-07-26:
I don't think that the branching feature of a version control system is the best way to separate different products.

Moorviper, 2020-07-26:
It is not a different product.
All printers are the i3.
you can also use mk3 parts on the mk2 or vice versa.

closed 2022-05-16 by Prusa-Support