prusa3d/Original-Prusa-i3 #148: Magnetic Bed Identification

beegmouse, 2020-05-04:
I had an idea for improving i3.

If some holes/slots were to be punched/cut outside the print area of the magnetic sheets, could the bed leveling sensor be used as a scanner and identify the bed during the calibration move sequence?

I frequently change between PEI and Textured sheets. 
They are different thicknesses and I have saved the presets.

It is easy enough to change the presets, but I wonder if it can be automated as I have forgotten to a few times resulting in failed prints or a scratched bed.

floridaservices, 2022-01-28:
Good to know I am not the only one who has problems with sheet-swapping

Prusa-Support, 2022-05-14:
Hello! I am sorry this hasn't been answered for a long time. I added the tag "enhancement" so that our devs can take a look and evaluate this in the future. No guarantees for now it would be implemented but with the tag I added it would now certainly be brought to their attention. 

Alessandro Pantaleo
Prusa Research