prusa3d/Original-Prusa-i3 #149: Hole in LCD Cover for MK2 is to tall

joseamirandavelez, 2020-05-13:
The hole for the LCD in the LCD cover for the i3 MK2 is too tall. I had to use the one for the MK2s. It fits perfectly in my MK2. Is that STL for a previous version of the LCD maybe?

https://github.com/prusa3d/Original-Prusa-i3/blob/MK2/Printed-Parts/stl/LCD_cover.stl

closed 2022-05-16 by Prusa-Support

joseamirandavelez, 2022-05-17:
It's been a couple of years. I don't remember exactly, but I think the rectangle height was too tall and I had to use the one for the MK2S.

joseamirandavelez, 2022-05-17:
I'm not at my computer right now. But it should be easy to check that by importing both STLs in the slicer.

Prusa-Support, 2022-05-18:
Yes, there are some minor design differences between the models. Most likely because our supplier of the LCD panels might have changed between the MK2 and MK2S versions.
The source files for the models are available so if anyone needs to do any custom changes they are welcome to. We have no plans to update the files in this repository as the MK2/s line of printers is no longer being sold.

Marek Micek
Prusa Research