prusa3d/Original-Prusa-i3 #150: Slic3r 1.3.0 not opening LCD_cover.stl

masquerain0601, 2020-06-15:
As i said in the title i can't open this file with Slic3r (not prusa Slic3r) it is in a loop, loading but it does not goes any further. I tried opening all other files and there are no issue

Moorviper, 2020-06-15:
And you are really sure it is a bug of prusa and the ticket belongs to this repo ?

BNoiZe, 2020-06-15:
I just tried to open the file in Slic3r myself. It is opening without problem. So I am certain this is a problem with Slic3r on your system. So I think this report belongs to the Slic3r repo.

Moorviper, 2020-06-15:
[image]
Yes works fine

masquerain0601, 2020-06-15:
Ok so it is a problem of my installation :/
Thank you

closed 2020-06-15 by masquerain0601