prusa3d/Original-Prusa-i3 #152: dxf file fails to open

erichstuder, 2020-08-01:
I try to open the dxf files located in the following location:
https://github.com/prusa3d/Original-Prusa-i3/tree/MK3S
I cloned the repository to my local machine.

I cannot not open the file MK3v8b.dxf with LibreCAD 2.1.3 on a Rasperry Pi 4.
LibreCAD just hangs. FreeCAD even crashes.
MK3-front-plate.dxf and MK3-rear-plate.dxf work fine with LibreCAD.

Could there be an inconsistency in the file?
Are there tools known to work with this file?

6ffm70, 2020-10-08:
@erichstuder That file can be imported successfully in my proprietary CAD, and in Inkscape.

Inkscape is an opensource 2D-vector editor that can also export DXF files. Maybe you try that one, edit the file and re-try in LibreCAD?

Prusa-Support, 2022-05-09:
This seems to be an issue specific to your system/hardware. Unfortunately, we are unable to help with this issue.

Marek Micek
Prusa Research

erichstuder, 2022-05-09:
I don't think it is system/hardware specific. I just tried it on new PC with Windows 10 installed.
There seems to be the same issue.
Also I haven't found an online viewer that works.

Could you please point out what tool you use?

Prusa-Support, 2022-05-13:
Hello! I tested it at this moment with Fusion 360 on PC and was able to open that file regularly, can you please attempt it that way? Let me know how it goes. Online viewer: we use this one: https://help.prusa3d.com/article/360-viewer_1877 

Alessandro Pantaleo
Prusa Research