prusa3d/Original-Prusa-i3 #153: einsy-base file missing in STEP

artur-panek, 2020-08-25:
would be great to have it ther because the .scad has some resizing issues

3d-gussner, 2020-08-25:
It is there https://www.github.com/prusa3d/Original-Prusa-i3/tree/MK3S/Printed-Parts%2FSTEP%2Fextruder-body.stp

You need to change the github branch for different models.

artur-panek, 2020-08-25:
Hi, you just showed me the extruder-body.stp and I'm looking for einsy-base.stp.

3d-gussner, 2020-08-25:
> Hi, you just showed me the extruder-body.stp and I'm looking for einsy-base.stp.

Sorry for wrong link.
Glad you found it.

3d-gussner, 2020-08-25:
Please close this issue.

artur-panek, 2020-08-25:
I haven't foud it yet.

closed 2021-03-29 by artur-panek

Exeu, 2021-04-06:
@3d-gussner @artur-panek and where it is now? im also looking for it...

bwees, 2022-02-11:
These files still do not exist. What is the reason for these files missing??