prusa3d/Original-Prusa-i3 #156: SCAD files missing from MK3S branch

gargan26, 2020-09-01:
Several SCAD files are missing from the MK3S branch.  In the spirit of open-source, I'd like to see all *original* source CAD files used to generate the most current production parts here. (I.E., SCAD, Solidworks, Inventor, etc...)

Has anybody spoken with Prusa directly about getting these source CAD files?

Moorviper, 2020-09-01:
You have to go through the branches to get all.
Or which one is missing ?

6ffm70, 2020-10-06:
I'd look here: https://github.com/prusa3d/Original-Prusa-i3/tree/MK3S/Printed-Parts/SCAD
Which ones are missing?

Prusa-Support, 2022-05-09:
As mentioned in #163 , if the part has been designed in OpenSCAD, then a SCAD file is provided. If it was designed in other modeling software, we provide STEP file, which can be imported and edited in many 3D modeling programs.

Marek Micek
Prusa Research

closed 2022-05-09 by Prusa-Support