prusa3d/Original-Prusa-i3 #158: Native Source Files (Autodesk Inventor)

gargan26, 2020-10-06:
Are we ever going to get access to the native source cad files (Autodesk Inventor, ...) for the printed parts of the MK3S?  It's easy enough to reverse engineer the parts that were originally designed using OpenSCAD into Inventor format, but for the parts that were designed with Inventor from the start (Extruder...), It's definitely a chore.  I would love to see the native source cad files included in this repo updated in sync with the official .STL files that are released by Prusa.

The extruder body is a prime example.  It is very difficult to determine how this part was designed, and I would love to see the design history for this part for my own knowledge about 3D printer design.

6ffm70, 2020-10-07:
@gargan26 I agree! Prusa Research already made this step (and even a bigger one) in the MINI repository, as outlined by _@EuphoricPenguin in https://github.com/prusa3d/Original-Prusa-MINI/issues/34#issuecomment-680317524_

Let me use this occasion to highlight my plea to support opensource development for mechanical parts too:
For lack of finding a standard for open collaborative mechanical engineering, I started a document to collect/compile what's important to community developers: 
[image]
This document is maintained here: [https://github.com/6ffm70/opensource_mechanicalengineering](url)

Native CAD files (centre bottom) are a vital part of mechanical development, but also other data as well as shown above. Of course, not everything is available/required in all projects, and other data might be missing from the illustration.

Prusa Pesearch published some specs, 2D drawings w/ tolerances, print settings and other useful data in the [MINI repo](https://github.com/prusa3d/Original-Prusa-MINI) already. Big thumbs up to P/R for this!
This, I'd love to see for the i3 too; most importantly for the Mk3s.

infosisio, 2020-10-14:
Prusa, we are waiting...

jhoblitt, 2020-12-07:
The MK3S+ commit has STEP files but no inventor files for the non-scad parts: https://github.com/prusa3d/Original-Prusa-i3/commit/d7f81f2ef2f313728b81def8b9385fcf47e366fe#

bergstar, 2021-01-05:
@josefprusa please follow up

Prusa-Support, 2022-05-16:
We are in the process of adding those files but it can take some time. Please bear with us.

Marek Micek
Prusa Research

bergstar, 2022-05-16:
> We are in the process of adding those files but it can take some time. Please bear with us.

How many years are you planning to spend on this issue?

EuphoricPenguin, 2022-05-21:
At least we'll all have noble white beards when the project files are released.

sl1pkn07, 2024-05-24:
2 years more in the count...

gregsaun, 2025-07-12:
I'm reopening this issue because the problem raised in the original post, which was confirmed as an error by your team 3.5 years ago, hasn't been fixed. A STEP file is a compiled format, not a native CAD file, so it cannot be used as a source file. You should either publish the native CAD files or remove any mention of the MK3s hardware being open source.

- See section "1. Documentation" of the open source hardware definition here :
https://oshwa.org/resources/open-source-hardware-definition/
- See section "Original Design Files" of the Best Practices For Open Source Hardware here :
https://oshwa.org/resources/sharing-best-practices/
- The Prusa Mini has the native CAD files but not the MK3s

bergstar, 2025-07-12:
Issue is not solved. I agree with @gregsaun