prusa3d/Original-Prusa-i3 #160: [Question] Why do some nozzles types have such low torque setting as 1.5Nm?

mop999x, 2020-11-23:
i am did some research due to nozzle leaks and was wondering prusa/e3d suggest to use 2.5-3Nm torque on nozzle to tighten it but other sources i found mostly for other nozzle types suggest torque of around 1.5NM here are few examples

https://docs.dyzedesign.com/nozzles.html#install-your-tungsten-carbide-nozzle

and i believe even 3d maker engendering says the same

do you know why that is?

i just wanted to know why tightening nozzle at 3NM that say 1.5Nm would cause a leak, if it would ?

[image]

tomavisar, 2020-12-07:
What does this have to do with this? Also it's because different materials have different levels of thermal expansion

Prusa-Support, 2022-05-10:
Hello! 
I am sorry for the late response to this thread. We can think of the nozzle as any bolt, so depending on the bolt at hand the manufacturer specifies a maximum torque to be applied. If we apply too much torque, we'd have a risk of damaging, deforming, stripping the bolt, which can result in a loose joint. Leaking filament from above the heater block is a result of a loose joint between nozzle and heatbreak. 

So in this case, the too much torque could have caused deformation, stripping, which might have led to a poor joint nozzle-heatbreak which causes the leaking. I'd recommend checking if also using the correct torque spec would not solve the issue - in that case, the loose joint would be likely caused, provided that nozzle and heatbreak are compatible, by a badly positioned heatbreak (can be inadvertently raised during the nozzle change procedure). 

Alessandro Pantaleo
Prusa Research

closed 2022-05-10 by Prusa-Support