prusa3d/Original-Prusa-i3 #161: Some STL file corrupted?

roberteando, 2020-12-12:
HI,

Apparently some of the files here "Original-Prusa-i3/Printed-Parts/STL/" are corrupted, there is no donwload button , and trying to dowload direcltly from the link, it obtain 111kb corrupted files.

The ones that I noticed are these ones:

x-carriage-back.stl
x-end-idler.stl
z-axis-top.stl

There is an alternative path to dowload it? could you fix it?

Many thanks in advance.

ulab, 2021-02-12:
I just downloaded [x-carriage-back.stl](https://raw.githubusercontent.com/prusa3d/Original-Prusa-i3/MK3S/Printed-Parts/STL/x-carriage-back.stl), [x-end-idler.stl](https://raw.githubusercontent.com/prusa3d/Original-Prusa-i3/MK3S/Printed-Parts/STL/x-end-idler.stl) and [z-axis-top.stl](https://raw.githubusercontent.com/prusa3d/Original-Prusa-i3/MK3S/Printed-Parts/STL/z-axis-top.stl) without any issues. 

You need to open the "Raw" version of the file and save them from their git page. For example for the [z-axis-top.stl](https://github.com/prusa3d/Original-Prusa-i3/blob/MK3S/Printed-Parts/STL/z-axis-top.stl) use the "Raw" button and save the shown text file.

PrusaSlicer opens them just as any other STL file.

Prusa-Support, 2022-05-10:
@roberteando If you are still having issues with the download, please contact our support at info@prusa3d.com and we can send you the files you need.

Marek Micek
Prusa Research

closed 2022-05-10 by Prusa-Support