prusa3d/Original-Prusa-i3 #164: Extruder body part breaks repeatedly

tkapler, 2021-01-14:
I tried to print twice https://github.com/prusa3d/Original-Prusa-i3/blob/MK3S/Printed-Parts/STL/extruder-body.stl part and both times it broke on similar place (even once i printed it with other parts on the bed and second time separately). All the other parts were OK. I have used ABS instead of PETG, so that may be part of the reason but still i think there s probably some problem on this place and probably some reinforcement would help.
[image]

tkapler, 2021-01-15:
small update - tried for the 3rd time, same settings as the second, just used corex - the same problem in exact same spot as the image on the left. I am looking to the slicer, it seems it always happens at the layer 152 (16.2mm), where bridge infill changes the way it prints. I am now trying to add support enforcers around the area, so that there would't be so big stress.

stahlfabrik, 2021-01-15:
The layer adhesion seems to be lacking. Are you cooling the abs print? You should tune your settings for optimal layer strength. 

I failed printing the parts (years ago) in abs-t. When they were nice and accurate, layer adhesion was lacking. With no cooling they were strong but inaccurate in shape. 

So I printed in petg as recommended and did not look back

tkapler, 2021-01-15:
All other prints were without any problem, they are perfect (the photo exaggerates some asspects, but in real life they are really smoothe and nice. I have used both default Prusa settings and that presented in the 3MF
just the info of the 4th print - adding supports helped.

Panayiotis-git, 2021-01-15:
Perhaps increasing the perimeters at this area would help.

shawntexan, 2021-02-24:
do 100% infill to ensure strength and lasting there is a lot of heat to deal with. all my parts i print 100%

Moorviper, 2021-02-25:
more perimeters is more efficient than 100% infill...

steakknife, 2021-11-15:
Printed in black Prusa PETG included with the M3KS+ upgrade kit. The bed sensor holder outside failed to print correctly exactly as in the first picture on the LEFT side. 20% infill as recommended and added 5 mm brim. Printed on PEI-laminated flat steel sheet with adequate Layerneer Bed Weld. Retrying with 100% infill.

dougshell, 2022-08-21:
The same part failed in the same way on my end printing in ASA on a Ultistik textured bed (that apparently has such good adhesion I need to use glue according to PRUSA). I have no other adhesion issues with my printer and have been printing on it for almost 2 years.  The same part failed in the same way on a Voron that was for a PIF order for ASA Prusa parts.

Prusa-Support, 2022-09-02:
Thank you for reaching us.

This part was designed with PETG material in mind and tested with it. Materials like ABS/ASA have substantially different characteristics. PETG offers good dimensional accuracy while ABS/ASA are well-known for their infamous tendency to shrink and consequent poor first layer adhesion.

For instance, the prints of the main post may have failed because of a poor first-layer contact area of the detached PINDA mount piece [left] or poor layer-to-layer adhesion and dimensional stability [right]. This is based on a quick guess but the reason for the print failure could be a more complex mixture of factors.
- In the first case, you may want to review the first layer quality and the spring sheet [preparation according to the material](https://help.prusa3d.com/category/material-guide_220), or maybe add a [helper disk](https://help.prusa3d.com/article/shape-gallery_301453) at the bottom.
- In the second case, you may want to improve layer adhesion (rise the temperature, reduce the speed...), and in the case of ABS/ASA avoid thermal shocks as much as possible (warm ambient, weak or disabled print fan cooling...). Also, please notice the PINDA mount connection has sharp angles which probably won't help in case of strong shrinkage materials. 

Our factory printed parts used to be printed in ABS a few years ago. We have reason to believe that PETG is more suitable for structural parts, and we switched to PETG as soon as the textured sheet prototypes came out.

I don't really see a concern about the part design so, if all your questions are answered, I suggest closing this issue and getting in touch with our Customer Support for troubleshooting.
https://help.prusa3d.com/article/customer-support_2287

Michele Moramarco
Prusa Research

Prusa-Support, 2022-09-22:
I believe this issue can be closed now.

This may seem to be a problem of printing, that could be solved through a proper troubleshooting session, and it is not necessarily related to the printer design.
Please consider reaching our Customer Support for help with similar issues.

Thanks for understanding.

Michele Moramarco
Prusa Research

closed 2022-09-22 by Prusa-Support