prusa3d/Original-Prusa-i3 #167: Add BOM (Documentation)

flub9, 2021-02-04:
Quite simply put, add a simple BOM in the repository. Ideally specifying part, size, printable (y/n), quantity, etc.

This is an issue on the @josefprusa Prusa 3 repository that was opened in 2012 discussed through 2015 and never resolved.

Assembly instructions and pretty much everything else is easy to find, but for whatever reason, nobody I could find has a good BOM.

floridaservices, 2021-02-15:
I have always assumed this to be a business decision - open source does not have to mean give it all away.  That said, it's not hard to figure out from the assembly manual what is needed as many have done, so if this is indeed a business decision, it's not accomplishing much.

flub9, 2021-02-15:
Good insight, I'm working on figuring everything out from the manual and getting it into the spreadsheet, mainly I've been using the full cad autodesk viewer model.

scotthomasringa-debug, 2025-10-15:
[image]@151 problems