prusa3d/Original-Prusa-i3 #175: suggestion - automate live z

rredford, 2021-09-07:
Basically, have a touchoff pad. It is topped with steel, and are triggered by very light touch. Configure is much faster, since it first triggers PINDA on pad, then moves over and slowly lowers nozzle to hit it. then offset is known.

It retains live Z advantages, but makes live z much faster to configure.

Prusa-Support, 2022-05-09:
Not sure from your description what exactly you propose. MK3S/+ doesn't have any sensor that reacts to touch or that measures force. PINDA/SuperPINDA is an inductive sensor and is basically used to measure a distance from the steel sheet or from the calibration point.
Live Z must be done manually/visually.

Marek Micek
Prusa Research

rredford, 2022-05-10:
It's why it is a suggestion. :) It's an upgrade to automate live z setting, using only trigger-able metal plate. trigger pinda on it, then touch off nozzle on it. it now knows the exact difference of height between nozzle and pinda.