prusa3d/Original-Prusa-i3 #178: Error in "X-Carriage-back" R4 bearing clamp position

Devilscave, 2021-12-29:
I have despite lubrication plain bearings the problem that my X-slide on the sides, near the Z-slide begins to clamp. That's why I have already damaged several rods and bearings. The measure of the rods is exactly 53mm apart.
I have now searched for the error and viewed the files "X-Carriage.stl" and "X-Carriage-Back.stl" version R4. The file "X-Carriage" shows no deviation of the bearing dimensions. But in the file "X-Carriage-Back", the screw holes match, but not the bearing shells. If you put them together, the upper two bearings are pressed down. Which of course leads to a narrowing of the bearing positions. If I put both parts on each other and position them with the screws, I have deviations in the bearing shell positions. Could you check and confirm this? I have tested the R4 parts of the MK3s and the MK3s+ package's.

mikromCZ, 2022-02-19:
can you check if it is repaired in R7? there are some bearings mentioned in the comments

Prusa-Support, 2022-05-09:
@Devilscave Have you checked the latest R7 parts if you still have the same issue? The x-carriage and x-carriage-back parts were slightly modified to minimize any possible excessive pressure on the bearings. However, even with the new design, overtightening the screws in the x-carriage-back can still still create excessive pressure on the bearings which could then damage the rods in the long run. 

Marek Micek
Prusa Research

Devilscave, 2022-05-09:
It looks like the positioning of the half -shells in R7 has been significantly improved. But I also see that it was apparently narrowed. So it is very possible that a higher pressure on the camp could be exerted when screwing. But at least the positioning of the half -shells now looks very even. I will test it soon

Prusa-Support, 2022-05-14:
Hello @Devilscave, thank you for the response. Fair point about the possible higher pressure, in my experience I haven't noticed issues stemming from the new R7 part but let me know how the test goes, in case I can pass the feedback on to our devs. But still, even with the improved design, let's be absolutely careful on not over-tightening the bolts on the X-Carriage-Back in general! 

Alessandro Pantaleo
Prusa Research