prusa3d/Original-Prusa-i3 #181: Address violations of GPLv2

Kreyren, 2022-02-08:
**DISCLAIMER:** This is not a legal advise that would create client to lawyer relationship and is provided as-is with claim of no liability to the author.

---

The Free Hardware Designs 3D printing machines were enabled by Dr. Adrian Bowyer and the work on The RepRap project since 2004, yet Prusa Research, a.s. fail to provide an appropriate credit per requirements of the license used for The RepRap Machine:

>   1. You may copy and distribute verbatim copies of the Program's
source code as you receive it, in any medium, **provided that you
conspicuously and appropriately publish on each copy an appropriate
copyright notice** and disclaimer of warranty; keep intact all the
notices that refer to this License and to the absence of any warranty;
and give any other recipients of the Program a copy of this License
along with the Program. -- GPLv2

Please address this violation of GPLv2 by adding the missing copyright notices to the used parts in this project.

### Reconstructed history of hardware development

  * 13rd September 2006 - RepRap 0.2[1]
  * 9th February 2008 - RepRap 1.0 ("Darwin") [1]
  * 2nd October 2009 - RepRap 2.0 (Mendel) [1]
  * 9th November 2010 - Prusa Mendel [3]
  * 12nd September 2010 - Prusa Mendel iteration 2 [2]
  * 21st May 2012 - Prusa Mendel Iteration 3 / Prusa i3 (this project) [4]

### Note on credits

In the spirit of Free Hardware Designs development I recommend adding a credit for The RepRap project in the README files of all relevant printers e.g.

```
Made possible by The RepRap Machine by Dr. Adrian Bowyer and contributors to The RepRap project <https://reprap.org/>.
```

### References:

1. History of The RepRap Project - https://all3dp.com/history-of-the-reprap-project/
2. First commit of PrusaMendel - https://github.com/josefprusa/PrusaMendel/commit/6ed4480bd708121c18f67fc6903092e7dbaecb5f
3. First commit to the RepRap wiki about Prusa Mendel - https://reprap.org/mediawiki/index.php?title=Prusa_Mendel&diff=23221&oldid=23220
4. First commit to Prusa3 - https://github.com/josefprusa/Prusa3/commit/d3618a65684dae11b45e364b28529af3ea1782e5

Moorviper, 2022-02-08:
what's your point?
the license is there

https://github.com/prusa3d/Original-Prusa-i3/blob/MK3S/LICENSE

Kreyren, 2022-02-08:
> what's your point? the license is there -- @Moorviper (https://github.com/prusa3d/Original-Prusa-i3/issues/181#issuecomment-1033014978)

I am aware, but it doesn't credit the RepRap machine from which this project was forked from. So point being credit the RepRap project for this design in the code.

Moorviper, 2022-02-09:
That isn't required

Kreyren, 2022-02-09:
> That isn't required -- @Moorviper (https://github.com/prusa3d/Original-Prusa-i3/issues/181#issuecomment-1033536786)

> You may copy and distribute verbatim copies of the Program's
source code as you receive it, in any medium, **provided that you
conspicuously and appropriately publish on each copy an appropriate
copyright notice** and disclaimer of warranty; keep intact all the
notices that refer to this License and to the absence of any warranty;
and give any other recipients of the Program a copy of this License
along with the Program. -- GPLv2

Seems clear to me -> It is required

dougshell, 2022-08-21:
A direct disagreement is posted above. This seems like it warrants a acceptance or closing of the posted issue

Prusa-Support, 2022-09-17:
Thanks for the heads up.

Credits are stated in the printer's manual but we wouldn't mind adding more credits. We will consider doing that.
Please close the issue if you think your concern was addressed in a satisfactory way.

Michele Moramarco
Prusa Research

Kreyren, 2022-09-17:
@Prusa-Support Add credit to the mentioned people and projects until that is resolved i am not convinced that it would be appropriate to close the issue.

Kreyren, 2022-09-17:
@Prusa-Support Please be aware that even though you've added a lot of changes it's still provable beyond a reasonable doubt that Prusa i3 is a fork or RepRap 2.0 which was released under GPLv2 so you might not have the legal right to change the license on GPLv3 as all contributors didn't agree to the license change.

tracernz, 2022-09-17:
> RepRap 2.0 which was released under GPLv2 so you might not have the legal right to change the license on GPLv3 as all contributors didn't agree to the license change.

The upstream is also licensed GPL-v3 for many years?
https://reprap.org/wiki/RepRap_Firmware
[The oldest release on GH](https://github.com/Duet3D/RepRapFirmware/blob/v1.18.1/LICENCE)

Kreyren, 2022-09-17:
I was talking about the hardware as the wiki says that it's GPLv2 (and phrased to be GPLv2-only):

[image]

When asking RepRap it's not clear if it's GPLv2-only or GPLv2-or-later.

Where the development of Mendel seems to have been done by https://reprap.org/mediawiki/index.php?title=Mendel&dir=prev&action=history:
* TheOtherRob
* Adrianbowyer
* Sebastien Bailard
* Buzz
* Mellis
* MarcusWolschon
* Rhododendronbusch
* GregFrost
* Enen
* MarkSteward
* Piglop
* Robbar
* Ethan sherman
* Peer
* Unti1x
* Тони
* AgeingHippy
* Sean
* Corentin
* Casainho
* Buback
* RhysJones
* Spacexula
* Csro
* Okpail
* Tombo9999
* Glenn
* DavidCary
* Infovis
* Piranha
* Andrewq
* Dereckson
* KalleP
* DavidCary
* WilliamBoshi
* Glenn

There are also lot of contributors to the prusa mendel that were not mentioned:
https://reprap.org/mediawiki/index.php?title=Prusa_Mendel&action=history

Prusa-Support, 2022-09-20:
Noted, thanks.
I'm collecting relevant information and forwarding it to the competent team.

Michele Moramarco
Prusa Research

Prusa-Support, 2022-09-24:
It seems it's done.
https://github.com/prusa3d/Original-Prusa-i3/pull/185

Please allow me to close this issue.

Michele Moramarco
Prusa Research

closed 2022-09-24 by Prusa-Support