prusa3d/Original-Prusa-i3 #186: Missing .STEP Files

SpecVR, 2022-10-11:
I was attempting to modify some of the parts used in the MK3S/+ and noticed that multiple .step files were missing, including the z-axis top and bottom. I was wondering if it would be possible to somehow obtain these or get the repository updated. 

Thanks

Moorviper, 2022-10-11:
Thats because they never existed.
These parts were don in openscad and the files are there.

Prusa-Support, 2022-10-15:
That's correct.

This is also explained in the relevant article of the Help section.
https://help.prusa3d.com/article/printable-parts-for-original-prusa_1824

>Some parts were designed using OpenSCAD, where SCAD files are the project files you can edit. For that reason, for all parts designed with OpenSCAD, there are only SCAD files and no STEP-files available. This also includes metal parts.

I hope you don't mind me closing this issue.
Thanks for understanding.

Michele Moramarco
Prusa Research

closed 2022-10-15 by Prusa-Support