prusa3d/Original-Prusa-i3 #188: Missing X axis parts in the STEP fiels folder

padillla, 2022-10-21:
What the title says, there are no X-axis STP files, please include them.

Moorviper, 2022-10-21:
https://github.com/prusa3d/Original-Prusa-i3/tree/MK3S/Printed-Parts/SCAD

There is the source file

padillla, 2022-10-22:
I tried the SCAD files but they have errors when I open them in Open SCAD

Moorviper, 2022-11-01:
They also work here...

bjanders, 2024-01-14:
It's not just the X axis parts, but other parts as well. I can open them in OpenSCAD, but I was not able to figure out how to export to STEP from OpenSCAD. It would be nice if they were directly available as STEP files in the repo.

I tried to find the intentional reasons from other reported issues, but didn't yet find them.  If the related tickets are known, it would be helpful if they were linked here.