prusa3d/Original-Prusa-i3 #189: x-end-motor and  x-end-idler scad file no version number

easonjim, 2022-11-01:
Dear ,
   x-end-motor and  x-end-idler scad file no version numbe. 
  The stl file has the version number of R6, but the scad file has no version number.

Prusa-Support, 2022-11-08:
Thanks for reporting.
I'll make sure your comment reaches the competent team.

Michele Moramarco
Prusa Research

easonjim, 2022-11-09:
#190 
Dear @Prusa-Support ,
x-end-motor and x-end-idler scad file no version numbe.
The stl file has the version number of R6, I added the R6 version number to it.

Prusa-Support, 2022-11-23:
Thank you for your patience.

The issue should be solved by now, please check again.
Also https://github.com/prusa3d/Original-Prusa-i3/pull/190 was fulfilled.

This issue will be closed now.

Michele Moramarco
Prusa Research

closed 2022-11-23 by Prusa-Support

Jopie80, 2022-11-23:
Thank you for your help, the Einsy board didn't function well.

Met vriendelijke groet/ freundliche Grüßen/ best regards,
Hans Vogels
Eikenwal 18
6595 BA Ottersum
Netherlands

[fdefc083-28ec-4387-b570-5cdf37c6b953]