prusa3d/Original-Prusa-i3 #191: Prusa MK3S+ Z smooth rod is not flush with Z axis top parts

SuperNeko, 2022-11-19:
Hello! 
I have a question about Z axis smooth rod and Z top plastic part. I assembled my printer (Original Prusa i3 MK3S+) sometime ago, but only now I noticed that I am able to move Z smooth rod up and down relatively freely, because Z smooth rod is not flush with Z axis top parts.
[image]
Although it does not move in XY directions, it can be moved in Z axis directions.
I am sure that smooth rods are fully seated in both Z axis motor mounts.
I also inspected official 3D viewer model and it has the same thing.
https://drive.autodesk.com/de296501e/shares/SH9285eQTcf875d3c539d9640fa5f8e3afd8
[image]
I do not have any problems yet (and by googling this "problem", looks like no one has), **but I am generously interested, if there is any explanation why it is designed like this?** While it it does not impact use of printer, all other axis smooth rods are constrained in all axes.
Like X and Y smooth rod do not move in any direction.

For example, in Prusa Bear mod, Z axis smooth rod are flush with the Z axis top parts.
[image]

Prusa-Support, 2022-12-09:
Hello! The reason for it is to allow for very slight tolerances in the length of Z-axis smooth rods. 🙂 

Alessandro Pantaleo
Prusa Research

closed 2022-12-09 by Prusa-Support

mikromCZ, 2022-12-09:
Just put a M3 nut on the top of the rod before you mount the top part 😎