prusa3d/Original-Prusa-i3 #192: Hall sensor instead of IR filament sensor

emard, 2023-01-25:
Instead of IR sensor with a PCB, steel ball and mechanic arm
with 2 magnets in repulsion, filament sensor can be
simplified to 1 or 2 cylindrical magnets 5x5 mm in attraction
and Hall sensor in 3-pin plastic transistor package 4x3x1 mm.

no PCB, no ball, no mechanic arm no physical rig for magnets
to stay together - easy assembly.
Less moving parts = less force required =
less friction to the filament.

Hall sensor Honeywell SS495A or similar for 5V is directly wired
to printer's motherboard, no additional electronic parts.

This is my project I'm currently upgrading MK2S->MK2.5S
you can copy, remix or take sensor design as inspiration.
https://github.com/emard/filament-sensor

Prusa-Support, 2023-01-28:
Hello! Thank you for your idea and for sharing it here! We'll check together with devs if this can be of interest, we'll let you know in case of anything in this thread. 🙂 

Alessandro Pantaleo
Prusa Research

Roostaff, 2023-01-28:
Thank's for the reply!
and logo embossing as in 2.6 alpha1 for text would be nice too!!! in stl
format or what could be done!