prusa3d/Original-Prusa-i3 #193: Screen STL provided has different dimensions than what shipped w/ my MK3S+

seanmcne, 2023-02-07:
Hi, I noticed after printing the screen provided from github - the dimensions are different.  Is it possible to get the updated screen STL matching the MK3S+ file so I can print a replacement screen with my own filament? Thanks!

prusa3d-bb, 2023-02-08:
Hi Sean,

thank you for the message. All print files for the printer you can download here. 
Have a good day. 

S pozdravem/Best regards,

Viktor Olijnyk
Customer Support (CZ/ENG/RU/UKR)
—————————————————
PRUSA Research | Partyzánská 188/7a | 17000 Prague | CZ 00420 222 263 718 | SK 00421 220 570 305
Your feedback is important to us!

—————————————————

Our 24/7 LiveChat help is available in our e-shop at the right bottom corner. 

Troubleshooting tutorials here! Follow us on Twitter and Facebook! Check out our Blog! Have a look at our Service guides!

--- original message ---

--- end of original message ---

seanmcne, 2023-02-08:
I downloaded the LCD screen cover and printed it - the LCD screen cover here in github has different dimensions than what shipped with my printer. Is this expected?

seanmcne, 2023-02-08:
Thanks, I will re-measure and analyze the differences I've seen and let you know if I'm still finding inconsistencies.

closed 2023-02-08 by seanmcne