prusa3d/Original-Prusa-i3 #196: [Improvement] Fix extruder cables drooping and reduce strain.

MartijnGevaert, 2023-03-16:
## Issue
I opened a ticket here 
https://github.com/prusa3d/Prusa-Firmware/issues/4083
Which nicely demonstrates how the cable coming from the back of the extruder can droop below the nozzle. This means it can potentially knock off tall prints.
[image]
The firmware ticket reduces the odds of issues happening, but it only works for filament unload, and for filament unloads before max height is reached.
If you where to print a 210*210*210 cube, then the issue would persist.

## Suggested solution
A solution for this could be to have the cables coming out of the back of the extruder at an upward angle, something like 30 degrees.
This way the downward drooping of the cable could be compensated for, and only printed parts need to be modified.
The 30 degree bend itself shouldn't add any wear on the cables because it can be added before the zip-ties which hold the cables in place on the back of the extruder.

### Reduced cable strain bonus
The change would also likely reduce the strain on the cables coming out of the controller box, and the extruder, as they wouldn't have to go from vertical to horizontal, but vertical to 30 degrees instead.
You can see a worst case scenario in this image below from PCMag. This would be better if the cables came out of the extruder at an upward angle
[image]
The lower the extruder is, the higher the strain relieve. Which is lucky because I suspect 98% of 3D printing happens in the bottom half of the print volume.

### Concerns and side effects

#### Cable strain at max z-height
when the extruder is at max z-height, and on the right side of the printer, the cable could be placed under more strain, or less. Needs to be tested.

#### Cable max extendsion distance towards back of printer.
Extruder cables could stick out further towards the back of the printer since they don't need to make such a sharp bend. Although it still won't reach as far back as the bed heating cables, so no likely to be an issue.

tooshel, 2023-03-16:
This isn't why I choose to print this alternate electronics box (long story) but maybe this will help?

https://www.printables.com/model/1910-yz-case

I'm in the middle of printing it now so I can't say for sure it even works but it should help with the cable management.

stepikovo, 2023-03-28:
@MartijnGevaert your trapezoidal nut is upside down

Prusa-Support, 2023-04-28:
Thank you, also for the suggestion on the firmware side.

Honestly, I like the idea but I'm not much concerned by the low-left position. I would be more concerned by the cables pointing upwards, as they will touch the top of the frame.
This design concept should be tested thoroughly.

A firmly inserted [nylon guide](https://help.prusa3d.com/guide/5-e-axis-assembly_169235#171922) and a well tight [cable wrap](https://help.prusa3d.com/guide/8-electronics-assembly_174100#175658) should improve the cables support and reduce the risk of saggy cable collisions.

As a side note, the new MK4 design improves the cable guidance substantially. 🙂 

Thank you again for sharing your idea.

Michele Moramarco
Prusa Research

Pmac627, 2025-01-31:
This issue still exists with the MK4S. I have a tall print I am repeatedly running for prototyping, and the extruder cables are hitting the print. I've tried orienting the print differently, but it still gets hit at some point by the cables. I've had to compensate by using a lot more support that is difficult to remove from the print, but it's the only thing that keeps the print on the sheet long enough to finish.

Note: I purchased the printer pre-assembled and am far from an expert 3D printer user.

In the GIF below, you can see the cable resting on the print and sliding off.

[image]

Any and all suggestions are welcome.