prusa3d/Original-Prusa-i3 #203: MK4 Frame & STL Repositories missing

7s3, 2023-06-26:
The repositories containing the MK4 frame files and STL/SCAD/STEP files for the MK4 printed parts are missing.

PatB42, 2023-06-27:
sorry but any links ? can't find it on repositories!

trivalik, 2024-08-20:
I guess you can find these files only at https://www.printables.com/de/model/451501-mk4mk39-printable-parts

Prusa-Support, 2025-07-30:
That's right, and thanks for the link!
We are uploading printable files and their source file on Printables.

Not so long ago, we opened a new micro-website that will possibly help us keep track of the disclosed information about our open projects, and let users conveniently browse through the various disclosed sources, all in one place.
Direct links to printable files should also be listed in there.

https://www.prusa3d.com/page/open-source-at-prusa-research_236812/

Thanks for your interest and support!

Michele Moramarco
Prusa Resaerch