prusa3d/Original-Prusa-i3 #206: [FEATURE REQUEST] Screen position adjustment

shledge, 2023-09-30:
I noticed that my LCD cover slightly covers the bottom of the LCD itself, making seeing temperatures etc. a bit difficult unless standing up. They're basically too close to the edge. I printed a custom LCD cover with a larger aperture which helps somewhat, but its still a bit difficult to see information at the bottom of screen.

Would it be possible to adjust the X/Y position of the screen via settings? Similar to what you seen for overscan settings on old PS1/PS2 games. If not possible, then perhaps increase the padding of bottom of screen to make bottom information more visible?

Prusa-Support, 2023-09-30:
Thanks for the suggestion.

On a side note, if the screen placement doesn't look like this, please reach out to the Customer Support. 

[image]

Michele Moramarco
Prusa Research

Prusa-Support, 2023-10-11:
I've just learned that the xLCD printed frame is intentionally narrow to protect xLCD panel as much as possible.
On a side note, given the tight tolerance margin, we are planning to adjust the UI for a clearer view. 

Michele Moramarco
Prusa Research

closed 2023-10-11 by Prusa-Support