prusa3d/Original-Prusa-i3 #207: MK4 non printable parts files

markaabo, 2023-12-02:
As the title says, will the non printable parts files be shared, stuff like the hotend and motors would be great for making addons to the mk4

Hope to get a response :)

s462022, 2023-12-02:
Based on history I would say likely not from PR.  However third parties inevitably do this with enough fidelity to do what you are wanting.  There are multiple for mk3, for example, that I have used for many add-ons and replacements.  Check grabCAD.

bolsoncerrado, 2023-12-20:
Is that the type of "open source" everyone claims Prusa sticks to?

Prusa-Support, 2025-07-12:
Sorry for the late reaction.
The firmware repository is always running hot 😄 so some of these issues may have slipped away.
Thank you @ipatch for your kind intervention.

For direct help, please don't hesitate to contact our [Customer Support](https://help.prusa3d.com/article/customer-support_2287).
Our Support Team is informed about the source of the printable parts and wouldn't deny addressing basic inquiries related to the Open-Source side of our products. 

Not so long ago, we opened a new micro-website that will possibly help us keep track of the disclosed information about our open projects, and let users conveniently browse through the various disclosed sources, all in one place.

- **https://www.prusa3d.com/page/open-source-at-prusa-research_236812/** 

From now on, we are committed to updating this reference web page with all Open Source related documentation and so far, the website is looking promising IMO.

Thanks for your interest and support!

Michele Moramarco
Prusa Resaerch

closed 2025-07-12 by Prusa-Support