prusa3d/Original-Prusa-i3 #209: Remove use of reference to "open source" from the project

bergstar, 2025-07-07:
This project is in clear in violation of what is required by open source hardware. 5 years after issue has been opened, no source files are released for MK3s #158 as well as 30+ other issues open here asking for source files for MK3s and MK4.

Current fact - printer is NOT open source, since source files are NOT available. This is a joke. It's like green washing but for open source hardware. 

Its not only misleading to the customers, but simple morally disgusting to act in this way.

ipatch, 2025-07-08:
mk4 printable parts,

https://www.printables.com/model/451501-mk4mk39-printable-parts/files

bergstar, 2025-07-08:
> mk4 printable parts,

Thanks - but this discussion is about sharing the source files. .STL files are not source files – they're exported production files. You can't build upon STLs, nor learn or analyze from the original assembly.

Reproducible or, in this case, customer-repairable hardware is **not** open source hardware.

Prusa-Support, 2025-07-12:
Please check https://github.com/prusa3d/Original-Prusa-i3/issues/158#issuecomment-3065379257 and https://github.com/prusa3d/Original-Prusa-i3/issues/207#issuecomment-3065472777. 🙏

closed 2025-07-12 by Prusa-Support